nickcoutsos/keymap-editor #222: Feature request: Support for custom keys defined in included *.h files.

eZtaR1, 2024-05-13:
Many non-US users have to use [this workaround](https://github.com/zmkfirmware/zmk/issues/177#issuecomment-1306247821) to make our keymaps work as nice as possible. 

Would it be possible to include the import or custom defined keys like this, or is there a workaround? :)

nickcoutsos, 2024-05-13:
I don't plan on adding support for parsing arbitrary headers because of the [complexity it adds](https://github.com/nickcoutsos/keymap-editor/wiki/preprocessor-Support), but as far as locale-specific keycodes I think the better option is [zmk-locale-generator](https://github.com/joelspadin/zmk-locale-generator). Keymap Editor still won't parse the headers, but it will recognize them when included in your keymap and will add the keycodes/symbols to the UI (you're still responsible for actually adding the header file to your repo, though).

eZtaR1, 2024-05-14:
> I don't plan on adding support for parsing arbitrary headers because of the [complexity it adds](https://github.com/nickcoutsos/keymap-editor/wiki/preprocessor-Support), but as far as [...]

Thanks! I'll try it out :)