nickcoutsos/keymap-editor #223: Everything disappeared in my keymap editor.

elrikov, 2024-05-21:
I have been tinkering with the GUI the past couple of weeks and all of a sudden I go into the editor and nothing is showing. No layers, Combos, Macros, etc. 
It shows it is still linked to my Repository and adv360 but nothing in the GUI. I have tried reverting to a previous build and I am still getting the same error. 

If I click View keymap data I can see my Layers, Behaviors etc. but I am getting No keymap layers defined in the Keymap Editor. 

[image]
[image]

nickcoutsos, 2024-05-22:
I'm guessing you've mistakenly introduced a syntax error somewhere along the way. Looking at your screenshot it appears that you've added a `combos` node inside your `keymap` node (its children can only be layer nodes), and looking at your repository there are some preprocessor defines in the `keymap` node that [it won't like](https://github.com/nickcoutsos/keymap-editor/wiki/preprocessor-Support). My recommendation would be to move them elsewhere (or remove them altogether if you don't need them, because again Keymap Editor will not support them).

elrikov, 2024-05-22:
Thank you for the advice. This resolved the issue. 
I messed something up in my never ending quest to add a Macro using a Modifier such as Ctrl+K to type out Hello!
It seems like such an easy task but I will get back to it. Thanks again!

nickcoutsos, 2024-05-22:
A macro for that might look like:

```dts
    say_hello: say_hello {
      compatible = "zmk,behavior-macro";
      #binding-cells = <0>;
      bindings = <&kp LS(H) &kp E &kp L &kp L &kp O>;
      label = "SAY_HELLO";
    };
```

And in the editor:

[image]

That "Add String Sequence" button lets you type out a string of characters and have it auto-generate all of the `&kp ...` bindings for you.

To bind that to <kbd>Ctrl</kbd>+<kbd>K</kbd> you would need to define a Combo specifying the appropriate key positions for your keyboard and then specifying the `&say_hello` behavior in the binding.

closed 2024-06-04 by nickcoutsos