nickcoutsos/keymap-editor #224: Is it possible to have a preview mode of the created keymap?

Zurga, 2024-05-28:
When I am in the process of making a keymap I make assumptions about where keys need to be based om my daily typing. However, there is a disconnect between trying it out when I imagine a key doing something and having a text box to type in.

Would it be possible to implement this preview area to try out the currently edited layout.

Zurga, 2024-05-28:
Of course it would be a lot of work to emulate ZMK to support all the features, so for a first version it could we layer switching and normal keypresses. For determining which layout to compare against I would say the user can select two commits.

nickcoutsos, 2024-05-28:
That would be really really tough. Emulating ZMK is one thing, another would be having raw signal data from the keyboard (e.g., "switch 37 is being pressed" rather than "UP_ARROW is being pressed") which is impossible without a specific protocol.

It does sound like something ZMK Studio (development is getting underway now) might enable, though I don't know what kind of priority that feature has.

nickcoutsos, 2024-05-28:
> If we are comparing commit "x" and commit "y", we know that a pressed key
is mapped to "a" in commit "x", then we can (hopefully) find the
corresponding key (say 39) in the matrix (assuming that we are starting on
the base layer). Then the output key can be whatever is mapped in commit
"y" to key 39.

But you can only do that for keys bound to `&kp` or similar behaviors, and only if they are unique in the keymap (e.g., if you have different keys bound to the same character on another layer for convenience, like symbols, there's no way to determine which you'd pressed). You can't detect anything like layer changes or activation of sticky keys, etc, without having a way to directly communicate with ZMK.

Zurga, 2024-05-29:
I was thinking that it might be possible to track layer changes and store
them in a stack based structure with the base layer as the first inserted.
If you have a momentary layer switch, you add the layer to the stack,
lookup the symbol in the layer and return it, after which you pop the stack.

When duplicate keys exist on a layer, the simplest rule would be to return
the first one.

My goal would not be to have a perfect simulation of my keymap, but a
rudimentary version where one can try out the most used things like &kp,
momentary layer switches, hold modifiers and others. The first version
could be only &kp with manual layer selection.

nickcoutsos, 2024-05-29:
> If you have a momentary layer switch, you add the layer to the stack,
lookup the symbol in the layer and return it, after which you pop the stack.

That's the main problem, my app won't know that.

Many behaviors, including the layer-change related ones, don't communicate anything to the OS when you're using them. They're not part of the keyboard HID spec, so they're processed entirely within ZMK along with other events to trigger alternate key presses. There isn't any existing functionality that would let Keymap Editor respond to a layer activating.

Even saying it doesn't have to be a perfect simulation is, in my opinion, really subjective, as different people will have different ideas of what the minimum requirements would be. If you are fine with testing simpler character-to-character mappings you might have better luck with something closer to the OS, or with something like Karabiner Elements for macos or (I think) AutoHotkey on Windows.

closed 2025-02-14 by Zurga