nickcoutsos/keymap-editor #225: Preserving parameters when changing behavior

sebmos, 2024-06-03:
I find myself switching between different behavior configurations for some keys, to narrow down precisely how I need them to work - I have 7 different hold-tap behaviors stored that I use for different keys. It would be neat, if the parameters could be preserved when changing between behaviors - at least if it's between the same behavior type, or if two behaviors share the same parameter types.

nickcoutsos, 2024-06-03:
So... once upon a time this actually was the case. I still remember how much effort I'd put into it back in the day, but since rewriting much of the interactivity for the newer binding dialog this preservation was lost.

I think it is more or less as simple as matching the old parameter types to the new parameter types, I'll just need to find the motivation to wade back in, look out for potential gotchas, and do the work.

Thanks for sharing this, especially with a video demo!

nickcoutsos, 2024-06-05:
Just pushed up a change for this. It wasn't terribly complex but I've only tested the simpler use cases (e.g. switching from `&kp` to `&kp` or `&hm` to `&mt` preserves parameters, switching `&mt` to `&lt` or something ridiculous like `&kp` to `&bt` doesn't). 

The murkier test cases would be around inferred behavior parameter like parameterized macros but I'm going to leave it as is for now.

sebmos, 2024-06-08:
It works - even better, it works for partial matches, too, i.e. moving from a Keycap/Keycap behaviour to a Layer/Keycap behaviour will preserve the second Keycap. I think you can close that one.

nickcoutsos, 2024-06-08:
Cool! I actually originally meant to disallow that, but mostly to deal with situations where the first parameter affects the type of the second parameter but it seems like a non-issue.

closed 2024-06-08 by nickcoutsos