nickcoutsos/keymap-editor #226: Focusing keycode filter when selecting second parameter

sebmos, 2024-06-03:
When changing hold-tap behavior settings, clicking on the first parameter will focus put the focus of the cursor into the "Available Keycodes" field. When I click the second one, that does not happen, which is a shame. I think the best behavior would be to not only put the focus on the field, but also select its contents, so I can immediately start typing to find the keycode I'm looking for. (Will users regularly want to continue from the first search?)

nickcoutsos, 2024-06-03:
Ooh. I wasn't sure about this initially but after reloading I got the video to play and I think I understand. Clicking onto the second parameter both triggers closing the active value picker panel and opening the new value picker panel.

In this case these value pickers happen to be the same component and I'd hazard a guess that this means React doesn't need to unmount and remount, which would mean the focus effect isn't being triggered as expected.

Will need to do some more investigation. I can't remember clearly if this is something I'd encountered before and found it too much trouble to resolve, but... hopefully it's not that bad.

Thanks for bringing this up!

sebmos, 2024-06-04:
You could force a remounting of the "Available Keycodes" React component by providing it with a `key` that changes depending on the parameter - might be the easiest way to solve this!

nickcoutsos, 2024-06-04:
That's my hope!

nickcoutsos, 2024-06-04:
Looks like that was it. I've made the change and pushed so this will be resolved now. While testing I noticed that the value picker isn't highlighting the currently selected value but apparently this regression was introduced some time ago so I'll investigate that separately.

closed 2024-06-04 by nickcoutsos