nickcoutsos/keymap-editor #227: Macro to hold Shift and press F1

Odaym, 2024-06-06:
Hi, I have some shortcuts for switching to different workspaces:
[image]

I wanted to make these as macros so that I could have the Hotkeys (1 and 2 for example) run them in 1 button.

I would like to know what I should be entering in these fields to achieve that, I'm really not sure what half of these fields mean and where to specify what I want to have, any help is appreciated, thanks a lot!

[image]

Odaym, 2024-06-06:
I think i need to stay in Macros and do this?

[image]

nickcoutsos, 2024-06-06:
That's the basic idea for the macro, but you'll also want to finish it off with `&macro_release` and another `&kp LSHFT` (or whichever shift you're using).

You may not necessarily need to do this as a macro, though. The more straightforward approach would be a bind like `&kp LS(F1)` which applies the modifier directly to the keycode. In the UI that would mean selecting the `F1` key code and applying the modifier like so:

[image]

I know that there are some cases where the OS or specific software (usually games) don't like this because the modifier is pressed and released in the same HID report but if that isn't the case here you can avoid the extra work of setting up a macro.

Odaym, 2024-06-06:
hi thanks for the response, ok but this means that now F1 will be automaticlly considered as Shift + F1, that still has me needing to press FN first, I want my Hotkeys in the Base layer (1, 2, 3, 4) to be pressed once and they can execute Shift + F1/2/3/4

Would that mean that after this step (modifier to F1 directly) then assign Hotkey 1 to F1 and that would be Shift + F1?

nickcoutsos, 2024-06-06:
These can be done independently; you can have a layer where the F-keys are unmodified and another layer where they are. In that case the modifier would only affect the keycode it's attached to.

Odaym, 2024-06-06:
lol I can just do the modifier + F1 on the Hotkey that I want, this literally should be all there is to it

[image]

closed 2024-06-06 by Odaym