nickcoutsos/keymap-editor #228: Experimental feature mouse emulation using out of date key codes

aaroncorgan, 2024-06-06:
It looks like mouse emulation development is moving along and the community has consoledated their efforts into a single pull request: [Mouse keys/button/scroll](https://github.com/zmkfirmware/zmk/pull/2027)

This PR is by pete johanson (creator of zmk) and combines the work of several other PRs and his own together and all of the authors are contributing to a single branch. The keycodes have changed and this has made the mouse emulation feature on the keymap editor no longer function and builds no longer compile until the user goes and manually edits their keymap to the new values.

Here is the documentation for the feature that details all the new keycodes: https://github.com/petejohanson/zmk/blob/feat/pointers-move-scroll/docs/docs/behaviors/mouse-emulation.md

In summary, these are the current codes and the values they need to be changed to:

| OLD | NEW |
|------|-------|
| *&mkp* | Unchanged |
| MB6 | <Removed> |
| MB7 | <Removed> |
| MB8 | <Removed> |
| *&mmv* | Unchanged |
| MOVE | <Removed> |
| MOVE_HOR | <Removed> |
| MOVE_VERT | <Removed> |
| *&mwh* | *&msc* |
| SCROLL | <Removed> |
| SCROLL_HOR | <Removed> |
| SCROLL_VERT | <Removed> |
| SCROLL_UP | MOVE_UP |
| SCROLL_DOWN | MOVE_DOWN |
| SCROLL_LEFT | MOVE_LEFT |
| SCROLL_RIGHT | MOVE_RIGHT |

These are going to be the final values for the eventual release, so there wont be any additional changes moving forwards.

nickcoutsos, 2024-06-07:
Yeah this is kind of what I wanted to avoid. It's good to know that the official implementation has nailed down its interface but my concern right now is that I have no insight into how many people are dropping other forks in favor of this one.

I think for everyone's sake what I'll do is add _all_ of the relevant behaviors/codes and make sure to make users aware of the differences.

Thank you for pointing me to this documentation, I haven't been able to keep up with a lot of these things lately!

nickcoutsos, 2024-06-07:
Pushed some changes and added a note to the wiki. It's not ideal but that's kind of the nature of this feature (trying to support things not in he main ZMK branch) anyway.

Feel free to re-open if something still isn't working.

closed 2024-06-07 by nickcoutsos