nickcoutsos/keymap-editor #229: Show label/node in Behaviors table

sebmos, 2024-06-08:
I have quite a few behaviours in my keymap, but the list just shows the type of behavior and the binding. It would be nice if this table could also show the label or node, to make it a bit easier to see what's what.

Even nicer, but likely harder to achieve, would be if it could show the number of usages and/or where they're used. (I have an inkling that one of them is actually unused, but last time I removed a supposedly unused one, I broke one of my layers. 🤦🏻)

[image]

nickcoutsos, 2024-06-08:
I like that idea! It shouldn't be _too_ difficult to add, I think. Even better might be to also break it down by layer/combo/macro/etc but that can come later on.

nickcoutsos, 2024-06-10:
I had a look today and seem to have reference counting working (not deployed), but I didn't actually absorb the first part of your idea. Do you have an example of that?

I wanted to build the editor with maximum flexibility (there _may_ be multiple labels, label(s) _may_ differ from the node name) but in practice I think most people just want to reference their behaviors by a consistent name which is why it defaults to keeping them in sync. Certainly if the values _are_ different it's good to be able to see them in the list but I'm imagining common use will tend to look like duplicate columns.

As a side note, when I refer to "label" here I mean in the devicetree sense (e.g. `label: node_name { ... };`) rather than the property as indicated in the app. I haven't gotten around to cleaning things up but for quite a while now ZMK has made `label` properties optional in behaviors/macros and even renamed layer labels to `display-name` so they don't serve much purpose anymore.

sebmos, 2024-06-11:
In my list, every one of those behaviors has a different label and node - for example, the first one `hra`, node `home-row-arrow`, label `Home Row Modifier (Arrow)`. Any of these pieces of information would be useful to show. I have no idea what `htb`, `hth` and `htt` do without clicking through to the config. 🙈 Pretty much any additional information would be useful.

nickcoutsos, 2024-06-11:
Ok, yeah, that'd be a problem for sure. While I don't love the idea of adding the label prop front-and-center in another way (I checked and it _is_ optional but it's also marked as deprecated\*), I could still just show it and/or the node name conditionally when present and sufficiently unique from the bind code.

\*This isn't the first time something has been deprecated in ZMK and there haven't been any plans I've seen to actually remove support for deprecated features (I suspect this may not happen before ZMK has versioning), but I'd still like to avoid giving a reason to continue using them.

sebmos, 2024-06-12:
I think the node name is perfectly fine and enough, especially if the label is deprecated. If users care about the readability of the list, they can edit/customise the node name. It does feel unnecessary to have both there, although it adds to the lack of ability to "describe" stuff in keymaps. But that tension is way beyond the scope of this feature request.

closed 2024-06-13 by sebmos