nickcoutsos/keymap-editor #230: Adding acute accent as keycode

Nikvothe, 2024-06-12:
Hi I love the software, far better than the one provided by Kinesis, but I would like the acute accent to be available as a keycode so that it would allow me to use it in combos easily.
It looks like this: ´

And it's usually used in vocals:

a --> á
e -> é
i -> í

nickcoutsos, 2024-06-12:
Hi @Nikvothe 

> I see it is not supported in macros either, the characters containing ´ are omitted. The rest of the string is fine though.

If you're referring to the "Add String Sequence" feature, that's a limitation in my app (and ZMK, but it doesn't help that I don't surface the unsupported characters).

Have you looked at [zmk-locale-generator](https://github.com/joelspadin/zmk-locale-generator)? I provides key mappings from other keymaps to QWERTY. It lets you define the keymap using familiar codes and symbols but still requires that your OS is configured to interpret the appropriate language. I _believe_ this includes accented characters.

If you can get the appropriate header file from that repo (under releases), add it to your repo under `config/` and include it in your keymap (e.g. `#include "keys_BLAH.h"` then the app will include the new keycodes.

Nikvothe, 2024-10-08:
Hi @nickcoutsos thanks for your indications.
It's been a while, but right around your response I found out my kinesis was faulty... It took the fun out of it for me but luckily I got a replacement. Now I'm trying to go for it again and run your software locally. Any clues on why $mo behaviour has no layers?

[image]

If you believe I should stick with your online endpoint I will and I will try to do something that way.
I'm trying to achieve a similar functionality that you can find on a mac when you combine option + e to get an acute accent, or when you combine option + n to get ñ. I wish to maintain those shortcuts, or something close to that, when I switch the keyboard over to a Windows computer.

nickcoutsos, 2024-10-08:
I don't update the source anymore so the version you can run locally doesn't have (among other things) dynamic options for those pickers.

As for the accents thing... tough question. It's out of scope of the app itself and as someone who isn't working in Windows (beyond playing games) I'm not aware of the best approach for this sort of thing. It looks like there are some equivalent shortcuts https://support.microsoft.com/en-us/office/keyboard-shortcuts-to-add-language-accent-marks-in-word-and-outlook-3801b103-6a8d-42a5-b8ba-fdc3774cfc76 you can use.

Nikvothe, 2024-10-13:
Hi @nickcoutsos don't worry it's fine I've learned lots fiddling with your software, your contribution is much appreciated.
And your online UI version kept me going for a long while, it allowed me to use the faulty keyboard by remapping some keys.
 
I was finally able to configure the working keyboard with ZMK using macros to map some easy combination of keys to alt + unicode, I can now use the US layout comfortably on Windows without having to switch to other languages. I can always add a layer and do the same for Linux if I ever need to.

closed 2024-10-13 by Nikvothe