nickcoutsos/keymap-editor #233: Feature Request: allow macros (or, ideally any behavior) in the quick/batch add

danielo515, 2024-06-22:
I want to take the opportunity to thank you again for this tool. It is amazing how an individual has created a better tool than all of the companies that make profit from selling keyboards offer to their customers. You have my respect sir.

The addition of the batch/quick edit was awesome, and it has already saved me time in recreating my existing keyboard layout (I lost my branch when switching laptops 😢 ).
However, being it limited to just "real" keys is, well, very limited. I have scenarios where I want to assign macros to many keys, or that I want to assign none to several keys in a row... the list can grow easily as I keep thinking about it.
I understand this may not be easy to translate to UI, but it doesn't have to be beautiful. Just like you have the QWERTY on the right, you can put the macros and other behaviors there.

Also, I noticed that when you create a new layer there is an icon to access this quick mode, but when the layer is defined it is hidden. I don't see why such UI change is needed, and I think it will be more consistent and easier to reach if it is always there.

Thank you again.

nickcoutsos, 2024-06-22:
Hey, thanks for sharing this. There's a lot to go over, and while I may not end up making all of the changes suggested here I hope I can give some explanation for the current implementations.

I'll start with this one because its simplest:

> Also, I noticed that when you create a new layer there is an icon to access this quick mode, but when the layer is defined it is hidden. I don't see why such UI change is needed, and I think it will be more consistent and easier to reach if it is always there.

The _Quick Assign Layer_ dialog was originally accessed from the layer's three-dot-menu along with other actions to make mass changes to layer bindings. I added that visual cue after the fact to point people at the new feature: anytime you're looking at a layer with (if I recall correctly) > 95% `&trans`/`&none` bindings it will prompt you to use quick assign in case you need to fill that layer with many of the same type of binds (ie, keypress or similar).

I've also added an indicator for layers referenced in conditional layer configurations and as you can imagine this sort of thing could get out of hand quickly so I prefer to keep things tucked away. Honestly, I'm not even sure I like having that prompt but the goal is to help with feature discovery.

> However, being it limited to just "real" keys is, well, very limited. I have scenarios where I want to assign macros to many keys, or that I want to assign none to several keys in a row... the list can grow easily as I keep thinking about it.

I know this probably isn't what you're asking for but it should (unless I'm misremembering a lot about this feature) allow you to select any behavior (including parameterized macros) that take a keycode parameter, with the goal of easily setting up a base layer just by typing. This takes care of the very first layer, as well as alternative colemak/dvorak/etc keymapping layers, or layers with alternate functionality like autoshift.

> I understand this may not be easy to translate to UI, but it doesn't have to be beautiful. Just like you have the QWERTY on the right, you can put the macros and other behaviors there.

Yeah, the key palette is probably, uh, _key_ to all of this. It seems like it's a core feature from the likes of Vial/Oryx/whatever (spoiler alert: I've literally never used a keyboard that _could_ use this software so I've never been able to try it out myself) and once I included it in the quick assign dialog it seemed weird not to support it more fundamentally in the app.

I think there's probably enough demand (quiet or otherwise) for a palette that it's worth trying to build. I've been thinking about it for a while, but life events that I don't want to get into have affected my ability/motivation to tackle a fairly large and sweeping change like this feature. Things may soon be changing again, but I don't want to speak too soon as there's the possibility I may have even _less_ time.

That felt like a lot of rambling so I'll stop there. Feel free to share your thoughts on things, I'm happy to keep discussing and maybe there's a simpler smaller version of this that can still help with productivity in the shorter term.

danielo515, 2024-06-24:
> The _Quick Assign Layer_ dialog was originally accessed from the layer's three-dot-menu along with other actions to make mass changes to layer bindings. I added that visual cue after the fact to [...]

I think that was a UX and UI improvement. I understand why you only show in that situation, and the fact that I prefer to see it all the time is a matter of preference, so no problem there.

> I know this probably isn't what you're asking for but it should (unless I'm misremembering a lot about this feature) allow you to select any behavior (including parameterized macros) that take a [...]

Can you point me to how? I don't see any way to pick, for example, one of my macros:
[image]

> I think there's probably enough demand (quiet or otherwise) for a palette that it's worth trying to build. I've been thinking about it for a while, but life events that I don't want to get into have [...]

It's not that it's going to save me hundreds of hours, not even a couple of dozens, so focus on your life, because I'm pretty sure implementing this will take a couple of hours. If it is useful to you, or you feel like adding it, yeah, but don't do it for a random request from some guy. I can live without it 😄 
In fact, I added as  much as I could with the quick editor, then finished on VSCode using search and replace.

nickcoutsos, 2024-06-25:
That's the dropdown I'm referring to; in theory it should be populated with any behaviors which accept a single keycode parameter. Any non-parameterized macro, or any behavior that accepts 0 or 2 parameters is ruled out.

Could you point me to one of your macros as an example?

The use case I'm talking about is something like [this autoshift macro](https://github.com/nickcoutsos/keymap-editor/wiki/Autoshift-using-ZMK-behaviors#firstly-a-parameterized-macro) where it accepts a single keycode param and keyboard input is the easiest way to provide that in a repetitive manner. In the case of macros that don't take parameters most likely we'd want the palette UI to provide an exhaustive (but organized) listing.

danielo515, 2024-06-27:
All my macros are sequences of strings, so they are not properly detected.
And, in the case I wanted to use it was not even that, it was just a quick
release of a key + a layer change

nickcoutsos, 2024-06-28:
Ok, yeah, that's what I expected. It'd require a palette to pick from.

danielo515, 2024-06-28:
Exactly!