nickcoutsos/keymap-editor #237: Nested behaviour usage not included in usage count

sebmos, 2024-07-06:
The "Uses" column on the "Custom Behaviors" page does not include usages inside other behaviours, e.g. in mod-morph behaviours.

nickcoutsos, 2024-07-06:
Thanks for the heads up I'll have to take a look at that. It should be checking references everywhere, including behaviors, but I may not have tested that exhaustively.

sebmos, 2024-07-15:
If it helps, the specific scenario I've run into this is with a mod-morph behaviour that references a hold-tap behaviour as its "normal" binding. Possibly related: the hold-tap behaviour is defined before the mod-morph behaviour.

nickcoutsos, 2024-07-15:
Sorry to have left this hanging, it ended up being a silly `map` vs `flatMap` mistake that only affected how I count references in behaviors. I did a quick test and it looks to be better now. Pushing changes momentarily.

sebmos, 2024-07-16:
Perfect, works now!

closed 2024-07-16 by sebmos