nickcoutsos/keymap-editor #238: "Missing devicetree root node." when using urob's zmk-helpers

k3d3, 2024-07-11:
I'd like to use urob's homerow mods - in their config, it is defined using zmk-helpers as:

```c
/* Homerow mods */

#define KEYS_L LT0 LT1 LT2 LT3 LT4 LM0 LM1 LM2 LM3 LM4 LB0 LB1 LB2 LB3 LB4  // left hand
#define KEYS_R RT0 RT1 RT2 RT3 RT4 RM0 RM1 RM2 RM3 RM4 RB0 RB1 RB2 RB3 RB4  // right hand
#define THUMBS LH2 LH1 LH0 RH0 RH1 RH2                                      // thumbs

#define MAKE_HRM(NAME, HOLD, TAP, TRIGGER_POS) \
    ZMK_HOLD_TAP(NAME, \
        flavor = "balanced"; \
        tapping-term-ms = <280>; \
        quick-tap-ms = <QUICK_TAP_MS>; \
        require-prior-idle-ms = <150>; \
        bindings = <HOLD>, <TAP>; \
        hold-trigger-key-positions = <TRIGGER_POS>; \
        hold-trigger-on-release; \
    )
MAKE_HRM(hml, &kp, &kp, KEYS_R THUMBS)  // left-hand HRMs
MAKE_HRM(hmr, &kp, &kp, KEYS_L THUMBS)  // right-hand HRMs
```

and then in the keymap, instead of e.g. `&kp F`, it's defined as `&hml LCTRL F` (and &hmr for the right hand).

As soon as I add the key to the keymap, keymap-editor gives me "Missing devicetree root node."

I assume it wouldn't be impossible to translate this to something that doesn't use zmk-helpers, but is this something that's of interest to support in the future?

nickcoutsos, 2024-07-11:
It's not going to work, things like urob's helpers and Miryoku are way too preprocessor-dependent to use with this app. I go into it in a little detail here: https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support

k3d3, 2024-07-11:
That makes sense. Thanks for the quick response!

closed 2024-07-11 by k3d3