nickcoutsos/keymap-editor #240: Missing encoder on right side of board, also encoders are "flipped"

ujl123, 2024-07-14:
2 questions

1. I'm unable to see the right side encoders in the UI. I see that a issue was raised here and was mentioned that the user should edit their lily58.json file (https://github.com/nickcoutsos/keymap-editor/issues/190), but looking at my repo, there is no .json file to edit. https://github.com/ujl123/zmk-config/tree/master/config 

2. Sort of off topic as this is a zmk issue more than your tool's issue, but my encoders seems to be flipped, probably due to the footprint. Is there a zmk issue to "flip" increment and decrement in the config/keymaps?

nickcoutsos, 2024-07-14:
The editor no longer automatically saves JSON layout data into the repo for common keyboards, but you can copy the `lily58.json` from [keymap-editor-contrib](https://github.com/nickcoutsos/keymap-editor-contrib/blob/main/keyboard-data/lily58.json) into your `config/` dir and edit there.

I'm not sure of the most correct handling of this, I haven't looked at it in a while but at the time I think none of the split keyboards enabled the peripheral side's encoder because they weren't supported in ZMK.

Sorry, I have no idea about flipped pinouts, I'm sure it's possible to override it somewhere but it doesn't involve this app or the layout metadata. You may want to reach out the PCB manufacturer if they provided your `zmk-config` repo or else check the ZMK Discord server for advice?

ujl123, 2024-07-15:
Thank you for the tips regarding how to find the .json file! I manually added two more sensors in the waterfowl.json and now I can see my two peripheral side encoders on your keymap-editor.

It was just a hail mary to see if your tool had a way to flip the cw/ccw . The more sane solution of course is to fix this in firmware. I'lll reach out to the vendor to see if they have a solution.

Thanks!

closed 2024-07-15 by ujl123