nickcoutsos/keymap-editor #241: missing Mouse move behavior

achtan, 2024-07-18:
Hi, I cloned this layout from Ben: https://github.com/achtan/zmk-config-from-ben
and there I can see the "Mouse move behavior" 
[image]

how every In my own repo: https://github.com/achtan/zmk-corne-5-col
I don't see that option...
How can I setup the Mouse move behavior?

nickcoutsos, 2024-07-18:
Mouse movement stuff is (I think) not yet merged into ZMK main so you'll need to enable it as an experimental feature (don't forget to update your `westl.yml` accordingly). See https://github.com/nickcoutsos/keymap-editor/wiki/Experimental-ZMK-Features#mouse-emulation

alehano, 2024-12-05:
> Mouse movement stuff is (I think) not yet merged into ZMK main so you'll need to enable it as an experimental feature (don't forget to update your `westl.yml` accordingly). See https://github.com/nickcoutsos/keymap-editor/wiki/Experimental-ZMK-Features#mouse-emulation

How exactly should I update west.yml to be able to use mouse feature?

nickcoutsos, 2024-12-05:
Your `config/west.yml` file should look something like:

```yaml
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
  projects:
    - name: zmk
      remote: zmkfirmware
      revision: main
      import: app/west.yml
  self:
    path: config
```

If you wanted to use the implementation in https://github.com/zmkfirmware/zmk/pull/778 for example, you might change it like so:

```patch
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
+    - name: krikun98
+    - url-base: https://github.com/krikun98
  projects:
    - name: zmk
-      remote: zmkfirmware
-      revision: main
+      remote: zmkfirmware
+      revision: mouse-pr
      import: app/west.yml
  self:
    path: config
```

fabricio-bastian, 2024-12-15:
Hi there! The mouse functionality is not merged to the main branch and there is official support 
https://zmk.dev/docs/keymaps/behaviors/mouse-emulation

It seems like the keymap output is including the incorrect file:
```c
#include <dt-bindings/zmk/mouse.h>
```
instead of:
```c
#include <dt-bindings/zmk/pointing.h>
```c

Mouse emulation also needs to be enabled via config:
```
CONFIG_ZMK_POINTING=y
```

Just a heads-up, since I just tried to update my config and then suddenly the mouse functionality stopped working :)

nickcoutsos, 2024-12-15:
Hmm, I'll try to look at this soon. The config aspect is annoying because I don't want to search out and parse those files...

nickcoutsos, 2024-12-16:
Ok, I've refreshed behaviors from the current ZMK source. Now when selecting a mouse behavior it'll show a little warning to ensure it's enabled in your build configuration. It's untested as I'm not around a ZMK keyboard but it should include the correct header and... I _think_ the binding parameters are unchanged.