nickcoutsos/keymap-editor #245: Include directive gets dropped when key layout is saved

NightWatchman, 2024-08-11:
In my repo I have the following:

```
#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/ext_power.h>
```

After using the tool it gets changed to:

```
#include <behaviors.dtsi>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/keys.h>
```

The keys include gets moved, which is ok, but the ext_power include gets deleted, which isn't.
Doesn't seem likely, but just in case it's relevant, the initial keymap had some encoder mappings which I deleted because my build doesn't have encoders.

nickcoutsos, 2024-08-11:
> The keys include gets moved, which is ok, but the ext_power include gets deleted, which isn't.

Why not? The editor is managing header include statements needed for behaviors you're using in the keymap. If you added some RGB behaviors it would add the appropriate header, and if you then remove the behavior it would delete the header as well. In this case you aren't using `&ext_power` as far as I can see so the header can be omitted.

NightWatchman, 2024-08-11:
Unusual build probably, but I'm using SuperMini NRF52840 controllers in a wired configuration with USB plugged into the left half of my lilly58 build, a TRRS cable between the halves and no batteries. When `#include <dt-bindings/zmk/ext_power.h>` is removed my keyboard stops working, but if I add it back in then it works fine. I haven't tried to figure out why. I just know it works when it's there, and it doesn't when it's not.

NightWatchman, 2024-08-11:
Thanks for the hint. I added an EP_TOGGLE key and it retains the header. Good enough for me.

closed 2024-08-11 by NightWatchman

nickcoutsos, 2024-08-11:
You mean the compiled firmware doesn't work? I didn't see that your build was failing when the header was removed, it looked like the jobs on those commits were all cancelled.

Not that I could feasibly help debug, but the header [just contains some preprocessor defines](https://github.com/zmkfirmware/zmk/blob/main/app/include/dt-bindings/zmk/ext_power.h) so I'm really surprised that would have any effect.

NightWatchman, 2024-08-12:
Yeah it compiled fine, but didn't work when flashed. My best guess is that without one or more of the defines, this hardware just defaults to only using power from the battery terminals and not the USB connector.