nickcoutsos/keymap-editor #254: Macro expansion

Jaxon-teBoekhorst, 2024-09-17:
When expanding macros from imported files comments also get expanded and break the config

nickcoutsos, 2024-09-17:
Could you provide an example? I'm not sure precisely what you're working with and how I might reproduce.

Jaxon-teBoekhorst, 2024-09-17:
I think macro is the wrong word, a line with "#define var val      // comment" will subtitle both the variable and comment, even mid line

nickcoutsos, 2024-09-17:
Yeah, the preprocessor's going to do that, it doesn't care about comments. Are you writing it out that way or did a comment line get merged onto a previous line after editing with the app? If you can share the original I might be able to track down the cause and correct it.

Just a heads-up, though, I [pretty much can't support](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support) preprocessor stuff so if you're relying on that to achieve more complicated keymaps there may not be a great path forward.

Jaxon-teBoekhorst, 2024-09-17:
I was converting from a hand written file, and have since fixed the problem, though though I would submit this as a possible suggestion to improve the experience of doing so

nickcoutsos, 2024-09-17:
Ah, ok, glad to hear you could resolve it. Unfortunately there's nothing I can do about preventing that as its written. If you've hand written a keymap with a line as you've described then the preprocessor is going to do the substitution regardless of whether you use the keymap editor.