nickcoutsos/keymap-editor #255: Github login is not working

ecambronerosaba, 2024-09-19:
[image]

When I go to the site this is what I get. I can't load any of my keymaps and relogging in doesn't work.

Sellith, 2024-09-19:
yup, got the same problem here, i couldn't save so I refreshed the page, and I couldn't link back to my github

nickcoutsos, 2024-09-19:
Thanks for the heads up!

Checking my server logs it seems that requests to GitHub's API are timing out. If my error logs are complete this started a little before you opened the issue and GitHub hasn't addressed it on their status page so I may have to poke around my host and see what's up.

nickcoutsos, 2024-09-19:
Well that answers that: https://status.digitalocean.com/incidents/jl76cw5xjq9c

nickcoutsos, 2024-09-19:
Ok, it should be resolved. I tested myself and keymaps are loading 👍

closed 2024-09-19 by ecambronerosaba