nickcoutsos/keymap-editor #256: Link to artifacts takes me to workflow run summary of a wrong workflow.

jfujitani, 2024-09-21:
I have 2 github workflows setup, 1 to build the firmware after a commit and another to render an image of the layout based on the keymap.  After I added the second to the workflow the link on the layout editor page routes me to the later instead of the workflow results that have the firmware artifacts.

nickcoutsos, 2024-09-21:
Hmm... I haven't looked at the workflow integration in a while but as I recall it's just getting the most recent worfklow run for the repo (and maybe branch?) because the action could be anything and I don't want to try to inspect the workflow config to see what it's actually doing.

Not _ideal_ but do you maybe have a way to control the order of workflow execution?

jfujitani, 2024-09-21:
Right now I just have both workflows trigger at the same time, I play around with chaining them to see if that helps.  Thanks for the reply.  Obviously this is just a minor thing that isn't going to be a showstopper.  Thanks for the great repo.

jfujitani, 2024-09-21:
I ended up just combining the workflows, not ideal from a separation of concerns standpoint but it's such a simple set of steps it works fine.

closed 2024-09-22 by jfujitani