nickcoutsos/keymap-editor #260: Feature: please support temper keyborad keymap

limichange, 2024-09-29:
## Photo
[image]

## Repo
https://github.com/raeedcho/temper

nickcoutsos, 2024-09-29:
I'm not home at the moment, but it looks like they've got PCB files in that repo. If you take the `.kicad_pcb` file and import it via https://nickcoutsos.github.io/keymap-layout-tools/ you can use the resulting layout data by saving it to `config/info.json` in your repo.

limichange, 2024-09-30:
🤩🤩🤩🤩🤩🤩
[image]

limichange, 2024-09-30:
yeeeeeee!
[image]

closed 2024-09-30 by limichange