nickcoutsos/keymap-editor #262: Feature request: Support for non-ZMK devices

vsisl, 2024-10-03:
Hi there!

**tldr: Would it be possible to add support for any keaboards (not just ZMK) to keymap-editor by taking advantage of [kanata](https://github.com/jtroo/kanata) (or another similar tool)?**

I'm kind of a newbie in the custom keyboard world. For the last few years, I've been using macOS and [ukelele](https://software.sil.org/ukelele/) to create a custom keyboard layout. My layout is pretty specific and not possible to replicate with e.g. [Microsoft Keyboard Layout Creator (MSKLC)](https://www.microsoft.com/en-us/download/details.aspx?id=102134)

Now I would like to switch back to Windows and I don't want to learn an new keyboard layout just because switching a platform. Also I work exclusively on laptops and I don't want to use an addtional keyboard. I want to use the keyboard of my laptop.

I came across of [kanata](https://github.com/jtroo/kanata) which can be used to customize any keyboard. The software looks great but after many hours of unsuccessful attempts, I realized the configuration of kanata is too complex for me. 

Now I came across of your keymap editor, which looks super awesome and user friendly! I do realize it works only with devices with ZMK firmware though. 

**My question is: Would it be possible to add support for any keaboards (not just ZMK) to keymap-editor by taking advantage of [kanata](https://github.com/jtroo/kanata) (or another similar tool)?**

nickcoutsos, 2024-10-03:
It's an interesting thought but unlikely.

Way back when I started developing this as an editor for my QMK-based keyboard and for a while I could use it to edit either QMK or ZMK keymaps but it was really a worst-of-both-worlds situation. Being able to more generically support ZMK features meant focusing on handling the devicetree model of behaviors and bindings and even with QMK there isn't a translation between the two if you want more than just the trivial features.

With kanata... well this is my first time looking at it. It looks interesting but, again, wildly different

vsisl, 2024-10-04:
Hi @nickcoutsos 

Thanks a lot for sharing your thoughts! I completely understand your conclusion.

By chance do you know about any other key-mapping tool similar to kanata, that would also work with all keyboards but would be easier to configure?

nickcoutsos, 2024-10-04:
I'm not super familiar with this space, but I've used Karabiner Elements on macos and know if AutoHotKey for Windows.