nickcoutsos/keymap-editor #265: Macros deprecated property

ReFil, 2024-10-17:
Not really sure if this is an issue or just something to note as changing it may have compatibility problems.

When macros are generated they include the property `label=<macro name in all caps>`. The label property has since been deprecated and can safely be removed, there's now another optional property titled 'display-name' that gets used e.g. in ZMK Studio UI. 

Removing the label property will break very old builds (I believe builds prior to the Zephyr 3.5 update), adding the display name property may break builds from the studio merge back. Perhaps a checkbox could be added to add the display-name optionally? it's not causing any problems to leave the label there, it only generates a build warning

nickcoutsos, 2024-10-18:
Ah, yeah, I've been putting that off forever. The compatibility problem is pretty annoying... right now it's only supported in layers and only by preferring the `display-name` property over `label` if it's already being used (and that's per-layer).

I think doing it The Right Way would involve figuring out a ZMK version based on `config/west.yml`, except that ZMK doesn't explicitly have versions and I don't want to think about examining ZMK forks to figure out which properties to use.

The checkbox thing is probably the least headache for me, but I'd probably still need to touch on a number of things to make it happen.

ReFil, 2024-10-18:
Yeah, if you just check the revision in west.yml it'll be tricky to pull that information, it'll just have the fork name or `main`. The only way i could think of doing it is examining the zmk forks west.yml to determine what version of zephyr it pulls in (to detemine if `label` is deprecated) and then maybe looking for the studio data in the repo too, but as you say that's not practical and is full of too many edge cases. other solutions like precategorising some of the more popular forks just aren't scalable.
There's nothing really wrong with leaving it as it is right now, it's the most compatible solution as it is, especially given studio is currently in beta it might not be worth upending things now for the sake of it. It's just something i spotted and thought I should raise