nickcoutsos/keymap-editor #272: ZMK Studio Unlock key code.

max-hill-4, 2024-11-27:
&studio_unlock
https://zmk.dev/docs/keymaps/behaviors/studio-unlock

nickcoutsos, 2024-11-30:
Hi

I can potentially add this easily enough but you should be aware you'll likely face some more incompatibilities. I haven't been able to keep up with ZMK Studio development but what I did see seemed like it'd be non-trivial to support here. 

Notably, ZMK Studio needs to be built with a configured number of possible layers, behaviors, etc, which the editor isn't looking for. Another user pointed out that when building with ZMK Studio you can't use `zmk,matrix-transform` which can be a problem for people who want to manage things like keyboards with 5 vs 6 column layouts.

Also, because I don't integrate the Studio protocol to communicate with your keyboard there's no runtime keymap -> devicetree source path so this would be a one-way trip. You would probably have a simpler time manually adding that bind to your keymap and switching to ZMK Studio.

max-hill-4, 2024-12-04:
Thanks for the reply, I think the best solution is to manually add `CONFIG_ZMK_STUDIO_LOCKING=n` into conf, as it seems a hassle to manually edit keymap files.