nickcoutsos/keymap-editor #278: Feature Request: Add key-delete button "🚫" under both "Quick &kp Assignment" and key-assignment popup

petreza-github, 2025-01-06:
When there are a bunch of &trans keys on a layer and I select "convert &trans to &kp", the keys become &kp with the prohibition-sign/crossed-out-sing (🚫) as a temporary assignment.
I want the ability to convert an already assigned key to this state, so that I can later do "convert &kp 🚫 to &trans" or "convert &kp 🚫 to &none". 
The two places where this can be implemented is the initial popup on clicking a key and on the side of the keymap pallete that shows up after selecting "Quick &kp Assignment"
Thanks!

petreza-github, 2025-01-06:
... it might not be a bad idea to add a few more buttons on the side that do not require configuration, like '&trans' and '&none' buttons

nickcoutsos, 2025-01-07:
Hmmm...

I see the benefit. I think using quick assign isn't the right approach, it was specifically meant to facilitate quickly filling in a base layer for `&kp` (or a derivative of it) but it's true there isn't currently a non-repetitive way to _remove_ key binds.

The ideal would be to enable _selecting_ one or more keys separately from just popping up the bind editor dialog, but I'd need time to think about the semantics and actually do the implementation.

Another possibility is adding a context menu to the layer editor so you can right-click a key and see some alternative actions, but in practice that's still making the user alternate between two actions... faster than the bind editor but still annoying.

The quickest might be introducing a new "Remove bindings" mode which, as long as its active, replaces bindings on any selected key with `&trans`/`&none` (potentially with a dropdown for flexibility). I'm going to let this simmer for a bit as there may be some quirks but right now I think it's feasible.