nickcoutsos/keymap-editor #280: How do I add en and em dashes?

carbon-starlight, 2025-01-12:
As these are very frequent characters you may find in any book on any page, it's strange not to see them in character picker. How do I add a character that's not in picker?

nickcoutsos, 2025-01-12:
I don't think they're defined in the HID spec. If they're not listed in ZMK's documentation (See _[List of Keycodes](https://zmk.dev/docs/keymaps/list-of-keycodes)_) then it's not something you'll be able to select here.

Usually these kinds of things have some OS-specific ways to enter them. With macos, for example, <kbd>Opt</kbd>+<kbd>-</kbd> will produce `–` and that can be handled in the editor by binding `MINUS` and toggling the `LALT` modifier. Windows probably does something similar-ish -- `LALT` + some number pad sequence, in which case you'd have to define a macro behavior.

carbon-starlight, 2025-01-12:
@nickcoutsos Okay... I just discovered that there seems to be no way to enter Cyrillic either. Surely the supposed way to type on languages using Cyrillic script are not altcodes?

nickcoutsos, 2025-01-12:
No, but it's not exactly built into HID or ZMK either: ultimately your keyboard will send ANSI QWERTY keys and your OS will need to be configured to interpret them in the desired layout. To help keep things straight people use `zmk-locale-generator` which provides keycode aliases so that you can refer to the QWERTY-layout keycodes by a more familiar name.

It's come up a few times so I'm just going to refer you to a comment in a previous post: https://github.com/nickcoutsos/keymap-editor/discussions/203#discussioncomment-8907451

nickcoutsos, 2025-01-16:
That's going to need a macro. You can read up on them in ZMK's documentation: https://zmk.dev/docs/keymaps/behaviors/macros

That'll look something like:

```dts
        new_macro: new_macro {
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            bindings = <&kp LS(LC(U)) &kp N2 &kp N0 &kp N1 &kp N4 &kp RET>;
            label = "NEW_MACRO";
        };
```

In the app you can build it up key-by-key like:

[image]

Depending on your use-case (I think games usually) it may not accept the modifiers being sent with the same press-release as the key so if you need to break it apart you can use the `&macro_press`/`&macro_tap`/`&macro_release` directives:

```dts
        new_macro: new_macro {
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            bindings =
                <&macro_press>,
                <&kp LC(LSHFT)>,
                <&macro_tap>,
                <&kp U>,
                <&macro_release>,
                <&kp LC(LSHFT)>,
                <&macro_tap>,
                <&kp N2 &kp N0 &kp N1 &kp N4 &kp RET>;

            label = "NEW_MACRO";
        };
```

 
[image]