nickcoutsos/keymap-editor #282: Conflicting defines for AR_ARABIC_ALEF_HAMZA in zmk-locale-generator

RezwanArefin01, 2025-01-21:
Issue https://github.com/joelspadin/zmk-locale-generator/issues/10 is relevant to Keymap Editor as well.

Current workaround is to manually edit the `keys_ar.h` file to disambiguate two two conflicting defines, but then Keymap Editor won't recognize them due to it having its own dictionary.

[image]

nickcoutsos, 2025-01-22:
Thanks for the heads up! I'm away from home until the weekend but it'll be a quick metadata refresh and deploy. I'll update once that's done.

nickcoutsos, 2025-01-25:
I refreshed and re-deployed this morning, try reloading it in the app when you have a chance and let me know how it goes!

RezwanArefin01, 2025-01-25:
Working as expected!

[image]