nickcoutsos/keymap-editor #283: Mouse move vertical/horizontal use wrong syntax

B-urb, 2025-01-23:
Hey thanks for this great Editor,

I noticed when I use the &mmv behavior with vertical/horizontal movement where you have to specify a value, the syntax seems to be wrong and the build fails. The output is:  &mmv MOVE_Y -400, while the expected output according to the ZMK docs is  &mmv MOVE_Y(-400)

nickcoutsos, 2025-01-24:
Weird! I'll have a look this weekend, thanks for the heads up.

nickcoutsos, 2025-01-25:
I used the wrong syntax for function-type params in behaviors, I've pushed a fix 👍

nickcoutsos, 2025-02-03:
Oof, I didn't deploy the backend. I'll try and get that out in the evening.

nickcoutsos, 2025-02-04:
Just pushed, @B-urb, let me know if you give it another shot.

B-urb, 2025-02-04:
Works now, thank you!

closed 2025-02-04 by B-urb