nickcoutsos/keymap-editor #284: Combos are not loading

AlejandroSanchez90, 2025-01-25:
is this intended? i have some combos on my corne, which i cannot see in the editor, even the combos created directly in the editor are not loading after a refresh so no way for me to edit them (they are working fine when ported to my corne)

nickcoutsos, 2025-01-25:
Combos should work fine, if they're not appearing then something in the keymap isn't being parsed correctly. Most often this comes up with the use of preprocessor macros. If you can link me to a specific keymap I can investigate.

AlejandroSanchez90, 2025-01-25:
this is my config https://github.com/AlejandroSanchez90/corne_eyelash

i bought this corne from aliexpress i forked its repo and updated the key maps thats it

AlejandroSanchez90, 2025-01-25:
looks like its working now, i reverted then readded it and i can see them now

closed 2025-01-25 by AlejandroSanchez90

nickcoutsos, 2025-01-25:
Awesome, thanks for the update!