nickcoutsos/keymap-editor #285: &caps_word cancel on modifier press

fpswilly, 2025-01-27:
I've tried adding in the below code to my keymap manually to get caps_word to end on a press of CTRL (since I regularly run into issues with shortcuts when I click out of a text box and that leaves caps_word active).

```
&caps_word {
    /delete-property/ ignore-modifiers;
};
```

For some reason, this doesn't seem to be working (I am code illiterate - which is why your amazing keymap editor tool is a godsend!). It would be great to have some more customisation for the caps_word behaviour in your visual editor (for example, like the following PR that enables editing the cancel and continue bindings [#1451](https://github.com/zmkfirmware/zmk/pull/1451)).

nickcoutsos, 2025-01-27:
I'm not sure how exactly you're going about it, if you're willing to share your repo I could get a better idea.

Aside from there there are a few issues attempting this with keymap editor. The `ignore-modifiers` property isn't part of ZMK's caps word behavior, so you'd need to build from that PR's branch. If you already are you'll see that the editor doesn't automatically pick up the changes -- in the past I've added partial support for the mouse behaviors as they were getting a lot of use, I haven't followed this one so I don't know if it'll be merged in that form or kept as a module... this means you won't be able to view/change these new properties in the app, but if you set them manually they should be preserved when making other changes.

fpswilly, 2025-02-08:
Hi Nick,

I appreciate your support for a newbie. Sorry for the delay in getting back to you, I've been trying to understand your feedback - haha.

So I didn't realise there was more to applying the PR edit to my keyboard than copying that code snippet into my .keymap file. I have done some reading on this topic and trying to understand github better. From what you've said, I need to fork urob's repo and then edit my keymap from my newly created forked repo. Have I understood that correctly?

I have now tried to fork [urob's personal repo](https://github.com/urob/zmk-config), but there is no .json in the config folder so I can't seem to get a keymap up in your editor.

nickcoutsos, 2025-02-08:
Yeah, it's a bit more involved, here's the ZMK docs for building from other ZMK versions: https://zmk.dev/docs/features/modules#beta-testing

Still, my editor won't that link to dynamically find which behaviors/options are available so that won't help you in this case.

Regarding urob's repo specifically, the editor won't parse those keymaps because of the heavy use of preprocessor macros (and also the use of `zmk,physical-layout`).

What you could do, maybe, is use that repo and replace the contents of the appropriate keymap file with what you had before. Like I mentioned in the last post you still wouldn't be able to see the new `ignore-modifiers` option, but you'd be able to assign `&caps_word` and make other changes through the app without overwriting that.