nickcoutsos/keymap-editor #289: Custom key toggle behavior

dylanmmckay, 2025-02-11:
I'm trying to create a macro which toggles a key only if the key is off. It seems in the ZMK documentation this can be done by defining a new behavior which is the key toggle function with mode on: https://zmk.dev/docs/keymaps/behaviors/key-toggle

I do see that in the behavior update/override, I can change the mode for key toggle, but my understanding is that I wouldn't have access to all three modes (flip, on, and off) if I simply override the behavior.

Is this something I should currently be able to do that I'm just not seeing? If not, could this option be added?

Thank you!

nickcoutsos, 2025-02-11:
I seem to have left the `zmk,behavior-toggle-layer` option out of the list for new behaviors, I'll have to update that.

nickcoutsos, 2025-02-13:
Just pushed an update, have a look when you can and let me know if that's what you needed.

dylanmmckay, 2025-02-16:
Apologies for the delayed feedback! It works great! I cannot find any issues. Thank you!

closed 2025-02-16 by dylanmmckay