nickcoutsos/keymap-editor #292: Unnecessary #include <dt-bindings/zmk/pointing.h>

invrtd, 2025-03-03:
Using the editor with cause duplication of `#include <dt-bindings/zmk/pointing.h>` looks like as follows 
``` 
#include <behaviors.dtsi>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/outputs.h>
#include <dt-bindings/zmk/pointing.h>
```
changes to
``` 
#include <behaviors.dtsi>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/outputs.h>
#include <dt-bindings/zmk/pointing.h>
#include <dt-bindings/zmk/pointing.h>
#include <dt-bindings/zmk/pointing.h>
```

nickcoutsos, 2025-03-03:
I think I just forgot to handle duplicates for different behaviors that reference the same header (e.g. `&kp` and `&kt`). I'll try and find time tomorrow to test and push a fix.

nickcoutsos, 2025-03-03:
Just pushed an update, let me know if you have a chance to try it again.

TheHackerTMofQueens, 2025-03-09:
Hi I tried to update a keymap today that had the pointer thing in it and it still failed. I was using a default layout that I copied from somewhere else where the person had mouse action layer.

This was the error I got in both the left and right side: 

[image]

nickcoutsos, 2025-03-09:
@TheHackerTMofQueens that's unrelated, it means the version of ZMK you're building against doesn't have pointer support (probably a somewhat older branch).

invrtd, 2025-03-09:
Sorry for the late reply but it does look like it has been fixed. 

There was also a bit of an issue with it automatically changing the order I had things set to import which caused some build issues. I managed to get around it by changing the file structure a bit but figured I should also mention that just so you would know.

closed 2025-03-09 by invrtd