nickcoutsos/keymap-editor #293: Keymap editor fails to generate

Skosulor, 2025-03-03:
The keymap editor fails to autogenerate a layout with the error message:

 "Something went wrong with the attempted automatic layout generation for your keyboard and there's no automatic way to recover, but if you want to reach out for assistance maybe we can work together to resolve any technical issues or else craft a manual layout for your keyboard."

I suspect it fails to generate from the provided .dtsi file since if I remove it I get an error that the keymap editor is unable to locate that file. 

The layout file:
``` c
#include <dt-bindings/zmk/matrix_transform.h>
#include "lintilla-layouts.dtsi"

/ {
    chosen {
        zmk,physical-layout = &default_layout;
    };

    default_transform: keymap_transform_0 {
        compatible = "zmk,matrix-transform";
        columns = <12>;
        rows = <4>;
        //      SW6                              |                              SW6
        //  SW2 SW5 SW9 SW12 SW15 SW18           |           SW18 SW15 SW12 SW9 SW5 SW2
        //  SW1 SW4 SW8 SW11 SW14 SW17           |           SW17 SW14 SW11 SW8 SW4 SW1
        //      SW3 SW7 SW10 SW13 SW16           |           SW16 SW13 SW10 SW7 SW3
        //                        SW19 SW20 SW21 | SW21 SW20 SW19
        map = <
         RC(0,1)                                                                   RC(0,10)
 RC(1,0) RC(1,1) RC(0,2) RC(0,3) RC(0,4) RC(0,5)   RC(0,6) RC(0,7) RC(0,8) RC(0,9) RC(1,10) RC(1,11)
 RC(2,0) RC(2,1) RC(1,2) RC(1,3) RC(1,4) RC(1,5)   RC(1,6) RC(1,7) RC(1,8) RC(1,9) RC(2,10) RC(2,11)
         RC(3,1) RC(2,2) RC(2,3) RC(2,4) RC(2,5)   RC(2,6) RC(2,7) RC(2,8) RC(2,9) RC(3,10)
                         RC(3,3) RC(3,4) RC(3,5)   RC(3,6) RC(3,7) RC(3,8)
        >;
    };
};

```

Files inside the config directory:

```
.
├── boards
│   └── shields
│       ├── dongle_display
│       │   ├── CMakeLists.txt
│       │   ├── Kconfig.defconfig
│       │   ├── Kconfig.shield
│       │   ├── custom_status_screen.c
│       │   ├── custom_status_screen.h
│       │   ├── dongle_display.conf
│       │   ├── dongle_display.overlay
│       │   ├── dongle_display.zmk.yml
│       │   └── widgets
│       │       ├── battery_status.c
│       │       ├── battery_status.h
│       │       ├── bongo_cat.c
│       │       ├── bongo_cat.h
│       │       ├── bongo_cat_images.c
│       │       ├── layer_status.c
│       │       ├── layer_status.h
│       │       ├── modifiers.c
│       │       ├── modifiers.h
│       │       ├── modifiers_sym.c
│       │       ├── output_status.c
│       │       ├── output_status.h
│       │       └── output_status_sym.c
│       └── lintilla
│           ├── Kconfig.defconfig
│           ├── Kconfig.shield
│           ├── lintilla-layouts.dtsi
│           ├── lintilla.dtsi
│           ├── lintilla.keymap
│           ├── lintilla.zmk.yml
│           ├── lintilla_dongle.overlay
│           ├── lintilla_kscan.dtsi
│           ├── lintilla_left_central.overlay
│           ├── lintilla_left_peripheral.overlay
│           └── lintilla_right_peripheral.overlay
├── lintilla.conf
├── lintilla.keymap
└── west.yml
```

PS Thanks for an amazing tool, I have used it for two other boards and it worked like a charm! <3

nickcoutsos, 2025-03-03:
Not completely sure why it'd fail from that, if I try to import your DTSI into [keymap-layout-tools](https://nickcoutsos.github.io/keymap-layout-tools/) I get a usable layout:

```json
{
  "layouts": {
    "default_transform": {
      "layout": [
        { "row": 0, "col":  1, "x":  1, "y": 0 },
        { "row": 0, "col": 10, "x": 10, "y": 0 },
      
        { "row": 1, "col":  0, "x":  0, "y": 1 },
        { "row": 1, "col":  1, "x":  1, "y": 1 },
        { "row": 1, "col":  2, "x":  2, "y": 1 },
        { "row": 1, "col":  3, "x":  3, "y": 1 },
        { "row": 1, "col":  4, "x":  4, "y": 1 },
        { "row": 1, "col":  5, "x":  5, "y": 1 },
        { "row": 1, "col":  6, "x":  6, "y": 1 },
        { "row": 1, "col":  7, "x":  7, "y": 1 },
        { "row": 1, "col":  8, "x":  8, "y": 1 },
        { "row": 1, "col":  9, "x":  9, "y": 1 },
        { "row": 1, "col": 10, "x": 10, "y": 1 },
        { "row": 1, "col": 11, "x": 11, "y": 1 },
      
        { "row": 2, "col":  0, "x":  0, "y": 2 },
        { "row": 2, "col":  1, "x":  1, "y": 2 },
        { "row": 2, "col":  2, "x":  2, "y": 2 },
        { "row": 2, "col":  3, "x":  3, "y": 2 },
        { "row": 2, "col":  4, "x":  4, "y": 2 },
        { "row": 2, "col":  5, "x":  5, "y": 2 },
        { "row": 2, "col":  6, "x":  6, "y": 2 },
        { "row": 2, "col":  7, "x":  7, "y": 2 },
        { "row": 2, "col":  8, "x":  8, "y": 2 },
        { "row": 2, "col":  9, "x":  9, "y": 2 },
        { "row": 2, "col": 10, "x": 10, "y": 2 },
        { "row": 2, "col": 11, "x": 11, "y": 2 },
      
        { "row": 3, "col":  1, "x":  1, "y": 3 },
        { "row": 3, "col":  2, "x":  2, "y": 3 },
        { "row": 3, "col":  3, "x":  3, "y": 3 },
        { "row": 3, "col":  4, "x":  4, "y": 3 },
        { "row": 3, "col":  5, "x":  5, "y": 3 },
        { "row": 3, "col":  6, "x":  6, "y": 3 },
        { "row": 3, "col":  7, "x":  7, "y": 3 },
        { "row": 3, "col":  8, "x":  8, "y": 3 },
        { "row": 3, "col":  9, "x":  9, "y": 3 },
        { "row": 3, "col": 10, "x": 10, "y": 3 },
      
        { "row": 4, "col":  3, "x":  3, "y": 4 },
        { "row": 4, "col":  4, "x":  4, "y": 4 },
        { "row": 4, "col":  5, "x":  5, "y": 4 },
        { "row": 4, "col":  6, "x":  6, "y": 4 },
        { "row": 4, "col":  7, "x":  7, "y": 4 },
        { "row": 4, "col":  8, "x":  8, "y": 4 }
      ]
    }
  },
  "sensors": []
}
```

So worst case scenario maybe you can just paste that into `config/lintilla.json` unless there's something else going on.

> `zmk,physical-layout = &default_layout;`

Like ZMK Studio stuff, maybe? I haven't kept up with that but the physical layout details got pretty complicated but it should still pick up that matrix transform. If there's a repo you're willing to share it could help me understand what exactly it's unhappy about.

Skosulor, 2025-03-04:
Cool thanks. Here is the repo https://github.com/Skosulor/zmk-config-lintilla.

About the ZMK studio stuff, I don't know, once I went from QMK to ZMK I found this repo and never looked back so I have barley made any changes outside the editor.

I basically copied files from this repo: https://github.com/ctranstrum/lintilla/tree/zmk

nickcoutsos, 2025-03-04:
I took a quick peek at the metadata generation again, it doesn't like the lack of `zmk,matrix-transform = ...` in the chosen node. Unfortunately, you can't specify that _and `zmk,physical-layout = ...`.

I _think_ you can just work around it by copying that generated layout into your repo, but looking at your keymap I can tell you'll run into issues parsing it due to the preprocessor macros (and probably also those reserved layers).

If you want I can take a closer look later and help you remove those to get the keymap loaded.

Skosulor, 2025-03-04:
I could just change the `physical-layout` to `matrix-transform`? 

I will change the whole keymap anyway, my plan was to do that in the editor but if the current setup will create issues I will modify it in the code directly.

nickcoutsos, 2025-03-04:
You can make that specific change, but the net result would be generating the same layout I posted above.

The other roadblock is the preprocessor macros. I was going to suggest you could just remove all the layers and remake them in the app if you wanted to but it affects the behaviors as well so it's up to you if you want to be a little more surgical with manual changes.

Skosulor, 2025-03-05:
Yeah I'll do exactly that, thank you!

closed 2025-03-05 by Skosulor