nickcoutsos/keymap-editor #294: How to setup 'unsupported hardware' by ZMK for the visual editor.

fpswilly, 2025-03-10:
Hi Nick,

I am going to set up a Temper and am struggling to understand the software side.

When I built my Sofle it was very easy to set up the visual editor. I used the default ZMK shell script tool to build my repo, and then it just worked in your editor. Since the Temper isn't on the ZMK support list, I was going to start with [Raedcho's Temper firmware](https://github.com/raeedcho/temper-zmk-config) to ensure everything works properly, but I will then want to customise the keymap once I've gotten it working.

I have forked Raeedcho's repo to my own. But your nifty little tool couldn't find a keymap to work off. I tried to move every file from the 'boards/shields/temper/boards' directory into the 'config' directory and it seems that the visual editor can see a keymap after this. Although, it can't load the keymap in due to the paramatized macros (which I don't need). When I delete the contents of the keymap, thinking this would get rid of those macros and allow me to start with a blank slate, I get a different error 'Could not parse keymap. Missing devicetree root node.' To be honest, even if I got it working this way, I still don't know if the github action would be able to build the firmware since I've moved all the files around in the repo.

How would you advise setting up a new keymap I can build for the Temper (3x5+6)? Would it make more sense to pretend it was a 36key Corne and set it up that way from the default ZMK script tool?

Sorry, I know I'm asking for a lot of help, but I just really can't get my head around how this works and would appreciate a nudge in the right direction.

Many thanks!

nickcoutsos, 2025-03-10:
**First**

The main thing is that you need to have a keymap file under `config/temper.keymap`, because the app won't look anywhere else.

**Layout**

Once you have that you can get the app to generate the layout data if you move `boards/shields/temper/` to `config/boards/shields/temper/` _or_ generate one manually using [keymap-layout-tools](https://nickcoutsos.github.io/keymap-layout-tools/):

1. Click on _Import..._
2. Select `boards/shields/temper/temper.dtsi` in the file picker, or switch to _Text_ and paste the file's contents into that box
3. Click _Import_
4. Copy JSON into `config/temper.json`

That won't have the nice visual separation, column stagger, etc., though. I see you copied a Corne layout. It doesn't look like temper has separate 5-column and 6-column transforms, but you can cheat it by deleting the `default_layout` section from your current file and the app will just use the first layout definition it finds.

**Parse error**

Looking at the original repo, the only preprocessor stuff that would cause issues is the `# define ______ &trans`. If you do find and replace for `______` and `&trans` in your keymap that would allow it to load properly.

fpswilly, 2025-03-17:
Nick, you are amazing! Thanks for your advice, I've got it working and it's no mean feat to coach a code-illiterate guy like me through this. I couldn't work out the visual separation, but that doesn't bother me to be honest - I know where all the keys are and at least I can now programme it to suit my needs. What's more, it would appear that the Github build action is still working to create my firmware so I'm feeling confident that this will work for me.

Thanks again!