nickcoutsos/keymap-editor #295: default corne 5 col is throwing an error when I try to load into keymap editor

kiganoakuma, 2025-03-19:
Could not parse keymap
It looks like some bindings in your keymap are using custom pre-processor macros which isn't supported by the Keymap Editor. These macros make it difficult both to resolve bindings to behaviors and parameters, as well as to generate modified keymap code that is consistent with your original intentions.

Please either continue editing your keymap by hand with these helpers or if you would like to use this editor, rewrite the keymap to use ZMK's syntax directly.

If you are looking for help in getting up and running or want a better understanding of the issue, please feel free to reach out to me:

[Keymap Editor Issues](https://github.com/nickcoutsos/keymap-editor/issues)
[Discord - nickcoutsos#4351](https://discord.com/users/115507966555193345)

not sure what the issue is, I forked the repo & then tried to load in, have created any  macros or behaviors yet, not sure how to fix it

nickcoutsos, 2025-03-19:
Can you point me to the keymap source?

ChristiamB, 2025-03-29:
> No se pudo analizar el mapa de teclas. Parece que algunas asignaciones de teclas en tu mapa de teclas usan macros de preprocesador personalizadas, incompatibles con el Editor de Mapas de Teclas. Estas macros dificultan la resolución de las asignaciones a comportamientos y parámetros, así como la generación de código de mapa de teclas modificado que sea coherente con tus intenciones originales.
> 
> Continúe editando su mapa de teclas manualmente con estos ayudantes o, si desea utilizar este editor, reescriba el mapa de teclas para utilizar directamente la sintaxis de ZMK.
> 
> Si busca ayuda para comenzar a trabajar o desea comprender mejor el problema, no dude en comunicarse conmigo:
> 
> [Problemas con el editor de mapas de teclas ](https://github.com/nickcoutsos/keymap-editor/issues) [en Discord - nickcoutsos#4351](https://discord.com/users/115507966555193345)
> 
> No estoy seguro de cuál es el problema, bifurqué el repositorio y luego intenté cargarlo, aún no he creado macros ni comportamientos, pero no estoy seguro de cómo solucionarlo.

I have the same problem

nickcoutsos, 2025-03-29:
> I have the same problem

https://github.com/ChristiamB/corne_config/blob/f934610253cb59a6af623022825b4b5dbfac5ef7/config/corne.keymap#L40-L43

The app won't parse these macros. See [Preprocessor Support](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support) for more details. If you're only using it for auto shift and don't want to write it out each time you can implement it with a hold-tap + macro instead. See [Autoshift using ZMK behaviors](https://github.com/nickcoutsos/keymap-editor/wiki/Autoshift-using-ZMK-behaviors).

https://github.com/ChristiamB/corne_config/blob/f934610253cb59a6af623022825b4b5dbfac5ef7/config/corne.keymap#L15-L17

To render the keymap as a 5-column layout the app would be expecting `zmk,matrix-transform = &five_column_transform;`. This change and the "reserved" layouts relate to ZMK's progression towards ZMK Studio. If you can find an earlier version of your corne repo to clone from that might make things easier, alternately see if ZMK Studio has what you need.