nickcoutsos/keymap-editor #296: feature request: write unspecified keycode (or recognize modules?)

harmy-fgc, 2025-03-23:
heyo! 

I am using the localization module for quite a few symbols and numbers, which requires to change the &kp COMMA to &kp DE_COMMA, which makes it unrecognizeable within the editor, and just adds the "stop" or wrong sign (the red circle, u know what i mean) 

currently i do most of my keymap via your editor, then push it, reedit the headers for the keys, push it again and build it. 

Alternatively, could it be possible for the editor to recognize the module [locales](https://github.com/joelspadin/zmk-locales)
and thereby allowing to just eddit them within the editor? 

Haven't found a decent way of doing it so far

nickcoutsos, 2025-03-23:
Supporting modules would be a lot due to needing to know what the modules are actually providing beyond things like clear cut behavior definitions. I can't generically parse keycode definitions because in the end they just evaluate to integer values.

That said, there is some amount of special handling of locales. If you include the header file directly (you'd need to copy it into your config folder so it's `#include "keys_de.h"` instead of `#include <locale/keys_de.h>`) then the app will load the correct symbols and make them available in the keycode picker.

harmy-fgc, 2025-03-23:
hey! 

I greatly appreciate the extremely fast response! 
A friend also showed me that he had the correct symbols, he didn't know how, but seems like i do now! :D 

Now i just have to figure out the folder structure, because so far i have a keymap in config/keyboard.keymap and shield/keyboard/keyboard.keymap and it's a bit odd to work with, but i think that's rather a ZMK thing to figure out. 

While also skimming through the available information of the editor, i realized that i overlooked exactly the topic of the localized headers, and there is a whole explanation :D 

so thank you dearly for still taking the time, despite already having it available! 

appreciate it, have a good one, and your editor is damn lovely!

closed 2025-03-23 by harmy-fgc

harmy-fgc, 2025-03-25:
hi, sorry for re-hijacking my own post, but i finally understood the behaviors, managed the headers, the editor is absolutely amazing. 
Cheers mate!