nickcoutsos/keymap-editor #302: Idea to support layer #define

cyb3rmonk, 2025-04-24:
Hi,

I read the preprocessor wiki page, and completely understand why supporting preprocessor macros in general isn't possible. That being said, I think there is a way you could support layer name defines.

The idea would be to add in the top horizontal menu a new defibeable thing - lets call it "Numeric #defines" for the sake of argument. Each numeric define would be a simple label+number pair.  Then, in the layer input widget, numeric defines would listed along layer numbers. And then internally, `&mo 2` would be represented separately from `&mo MY_LAYER`.

Done that way, seems it solves the serialization/re-serialization problem becomes solvable. Does that make sense?

nickcoutsos, 2025-04-26:
I'm not sure I'm seeing the solution properly. The serialization of it would still be a preprocessor `#define` statement unless you mean that this is the user's identification of existing defines? Even then there's still the concern of synchronizing layer changes (addition/deletion/re-ordering).

nickcoutsos, 2025-04-26:
Okay, so in any scenario we have to start by parsing `#define name number`. Not impossible but we can trip up by people wrapping the statements in `#ifndef`/`#endif` or in another header file that's being `#include`'d.

Even if this part works out we need to track which of these identifiers the user meant to be a layer which means encoding this preference in the keymap file, persisting it in the web app's local storage, or throwing it into a JSON file (assuming the GitHub repo use case here). The keymap file option is the most portable, but not resilient to manual changes by the user -- I kind of have to assume this is a real possibility for users who are interested in reading keymap files this way.

> Another option could be to explicitly support only layer number #defines and to just assume all numeric defines refer to layer numbers.

If I were to do that it'd definitely need to be a user option, but it still feels pretty iffy.

I could just be biased but I don't know how much layer tweaking I'd have to be doing before the lack of layer identifiers frustrated me enough to do this work. What does your keymap editing workflow look like now?

cyb3rmonk, 2025-05-02:
What the current support for preprocessor directives? Are you preserving preprocessor lines right now?

I could be wrong because I don't know what your tool does and doesn't support, but seems like you're letting perfect get in the way of better. Does this idea break some existing supported use case? Seems like supporting a very limited subset of directives is better than nothing - especially if it's the most commonly used subset.

> What does your keymap editing workflow look like now?

I use your tool to edit my keymap, and do simple tweaks manually sometimes. It's just annoying to have to count layers to figure out what an index refers to so I wish it was labeled.

nickcoutsos, 2025-05-03:
It does its best to preserve any part of the keymap file it can't directly process. So `#define` lines should be left alone. Additionally, although I haven't tried it in a while, it may _attempt_ to resolve identifiers in keymap layers. So if you have `&mo LAYER_FOO` in your keymap it may resolve it to `&mo 1` in memory, keeping the original `&mo LAYER_FOO` bind if you haven't touched it in the app.