nickcoutsos/keymap-editor #304: It is not possible to input value "0" to fields `Tap (ms)` and `Wait (ms)`

wanyann, 2025-04-30:
I want to assign value "0" to fields `Tap (ms)` and `Wait (ms)` in macro, but lowest possible number is 1.

[image]

[image]

Use case:
I use non-qwerty layout, and layer 4 contains basic qwerty layout to use with shortcuts

[image]

[image]
Behavior `hrml` contains `ml` macro as hold action.

[image]

The idea is to use Parametrized Macro as a home row mod, so when i held the button, keyboard will simultaneously press desired modifier and change layer to 4. 
But, because of wait time of "1", home row mods are messed up. The keyboard isn't switching layers quickly enough.
Things happening like that:
1) I press J (when held Ctrl)
2) Quickly press B, S in qwerty layout, desired shortcut is Ctrl + S
Result: Ctrl + B registered.

I setted these value to zero in code, and now it works perfectly.
```
        ml: ml {
            compatible = "zmk,behavior-macro-one-param";
            #binding-cells = <1>;
            bindings =
                <&macro_param_1to1 &key_on MACRO_PLACEHOLDER &layer_on 4>,
                <&macro_pause_for_release>,
                <&macro_param_1to1 &key_off MACRO_PLACEHOLDER &layer_off 4>;
            label = "Mod Layer";
            wait-ms = <0>;
            tap-ms = <0>;
        };
```

So, if it works with 0, it would be nice to have ability to set 0 in keymap editor.

nickcoutsos, 2025-04-30:
I'll have to dig into this later in the evening. Quick guess is that I'm being dumb and doing a simple "truthy" check on the value which treats `0` as missing/empty. As long as ZMK accepts those values then I should be accepting them as well.

nickcoutsos, 2025-05-01:
It was as I expected, just pushed a fix.

wanyann, 2025-05-01:
Just tested it, now it works perfectly, thanks for quick fix!