nickcoutsos/keymap-editor #316: Cannot assign both BT_SEL and BT_DISC to a hold-tap

mkovaxx, 2025-07-04:
## Summary

I seem to hit an edge-case in the UI when trying to configure a hold-tap behavior with two Bluetooth actions that both take the index of a profile.

## Details

The keyboard I'm configuring has 5 Bluetooth profiles and 6 keys I can dedicate to BT. I want each of the first 5 keys to select a profile when tapped, and disconnect that profile when held.

I have the following hold-tap behavior configured:

[image]

However, when I try to assign that behavior to keys and pick the parameters, it seems to trigger an edge-case in the UI:

[image]

It is impossible to select e.g. `hold: BT_DISC 0` and also`tap: BT_SEL 0`, which is what I need. When I click the list item for the `TAP: 2. Bluetooth Action` part, it has no effect; the action remains undefined.

Am I going about this the wrong way, or is this a bug?

nickcoutsos, 2025-07-05:
So... this is a slight complication of how behaviors work. Hold-tap lets you select 2 behaviors and pass along 1 parameter each. The difficulty is that the bluetooth behavior actually takes two parameters (the `BT_SEL` and `BT_DISC` actions count!) so you can't assign another via hold-tap.

One workaround is parameterized macros. Using them you can select one parameter in the definition itself and then bind the other when assigning it to a key. I was just taking some screenshots to demonstrate how to set this up but it looks like there's a bug with the macro editor so I'm going to have to look into that later. I'll get back to you when I know more.

nickcoutsos, 2025-07-08:
Hey, just want to give you a quick update. I did some debugging on the weekend and I can see where it's failing to handle parameter substitution for macros into a `&bt` binding. It's been a while since I've looked at this, so I'm wondering why exactly the same issue isn't cropping up for similar usage of `&rgb_ug` and want to make sure I wouldn't be breaking anything with a simple fix.

I'll try to get some quiet time in the next evening or so, but if I can't then this weekend at least should be a little more free. In the meantime, if this is a blocker, it should still be possible to hand-write the behaviors and use them in the editor. Let me know if you're interested and I'll whip up some sample code.

nickcoutsos, 2025-07-14:
Finally had a chance to look into the parameter issue! You should be able to create a parameterized macro with `&macro_param_1to2` and `&bt BT_SEL` (and a separate macro with `&bt BT_DISC` instead), then create a custom hold-tap behavior using these two macros.

At this point you can bind to this new hold-tap behavior directly in the keymap, or clean it up a little more with an additional parameterized macro that maps a single input parameter to both bindings. There's a bit more to it than that but if you're interested I have an example implementation for [Autoshift using ZMK Behaviors](https://github.com/nickcoutsos/keymap-editor/wiki/Autoshift-using-ZMK-behaviors#again-parameterized-macros-can-simplify-this-so-well-create-one-more) that does something similar.