nickcoutsos/keymap-editor #326: Stopped working for me, CORS issue?

prenaux, 2025-08-07:
It was fine in the morning but later in the afternoon it stopped loading. I tried to login in a fresh browser and relinking with github but that didnt solve it. Tried in Chrome and Firefox on macOS and Windows.

Here's screencap:

[image]

All the console messages:
```
Uncaught (in promise) Error: Request aborted
    exports https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    onabort https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
vendors.fc7fc484474dee29ffb7.js:2:96808
SES The 'dateTaming' option is deprecated and does nothing. In the future specifying it will be an error. lockdown-install.js:1:194912
SES The 'mathTaming' option is deprecated and does nothing. In the future specifying it will be an error. lockdown-install.js:1:194912
XHROPTIONS
https://keymap-editor-api-m89m6.ondigitalocean.app/github/installation
CORS Missing Allow Origin

Source map error: Error: request failed with status 404
Resource URL: https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js
Source Map URL: vendors.fc7fc484474dee29ffb7.js.map
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://keymap-editor-api-m89m6.ondigitalocean.app/github/installation. (Reason: CORS header ‘Access-Control-Allow-Origin’ missing). Status code: 504.
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://keymap-editor-api-m89m6.ondigitalocean.app/github/installation. (Reason: CORS request did not succeed). Status code: (null).
Uncaught (in promise) Error: Network Error
    exports https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    onerror https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    exports https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    exports https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    exports https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    request https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    exports https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    e https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    p https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    w https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    _ https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    f https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    a https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    p https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    p https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    value https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    t https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    p https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    w https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    _ https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    f https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    a https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    p https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    p https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    value https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    Fe https://nickcoutsos.github.io/keymap-editor/GithubPicker.2598399af837a4c8ebcb.js:2
    ni https://nickcoutsos.github.io/keymap-editor/react.0b7d83a58560e2038899.js:2
    Ss https://nickcoutsos.github.io/keymap-editor/react.0b7d83a58560e2038899.js:2
    ws https://nickcoutsos.github.io/keymap-editor/react.0b7d83a58560e2038899.js:2
    B https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    S https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    60053 https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    b https://nickcoutsos.github.io/keymap-editor/runtime.631e3f6d5c70f5449939.js:1
    63840 https://nickcoutsos.github.io/keymap-editor/vendors.fc7fc484474dee29ffb7.js:2
    b https://nickcoutsos.github.io/keymap-editor/runtime.631e3f6d5c70f5449939.js:1
    64448 https://nickcoutsos.github.io/keymap-editor/react.0b7d83a58560e2038899.js:2
    b https://nickcoutsos.github.io/keymap-editor/runtime.631e3f6d5c70f5449939.js:1
    73935 https://nickcoutsos.github.io/keymap-editor/react.0b7d83a58560e2038899.js:2
    b https://nickcoutsos.github.io/keymap-editor/runtime.631e3f6d5c70f5449939.js:1
    20745 https://nickcoutsos.github.io/keymap-editor/react.0b7d83a58560e2038899.js:2
    b https://nickcoutsos.github.io/keymap-editor/runtime.631e3f6d5c70f5449939.js:1
    24156 https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    b https://nickcoutsos.github.io/keymap-editor/runtime.631e3f6d5c70f5449939.js:1
    <anonymous> https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    O https://nickcoutsos.github.io/keymap-editor/runtime.631e3f6d5c70f5449939.js:1
    <anonymous> https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
    a https://nickcoutsos.github.io/keymap-editor/runtime.631e3f6d5c70f5449939.js:1
    <anonymous> https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js:2
vendors.fc7fc484474dee29ffb7.js:2:96808
Source map error: Error: request failed with status 404
Resource URL: https://nickcoutsos.github.io/keymap-editor/react.0b7d83a58560e2038899.js
Source Map URL: react.0b7d83a58560e2038899.js.map
Source map error: Error: request failed with status 404
Resource URL: https://nickcoutsos.github.io/keymap-editor/GithubPicker.2598399af837a4c8ebcb.js
Source Map URL: GithubPicker.2598399af837a4c8ebcb.js.map
Source map error: Error: request failed with status 404
Resource URL: https://nickcoutsos.github.io/keymap-editor/runtime.631e3f6d5c70f5449939.js
Source Map URL: runtime.631e3f6d5c70f5449939.js.map
Source map error: Error: request failed with status 404
Resource URL: https://nickcoutsos.github.io/keymap-editor/main.fe2b6919e0368e85cf5e.js
Source Map URL: main.fe2b6919e0368e85cf5e.js.map
```

prenaux, 2025-08-07:
Ok, well it seems to be back but I got some odd "Cant save" errors before it was fully back on. Looked like some DO server was struggling. Anyway closing for now since it works again, it was out for 2-3hrs maybe.

nickcoutsos, 2025-08-07:
That's... weird. Sounds like it was an outage of some kind but I don't see any incidents with DigitalOcean or GitHub.